Georacer/ardupilog #3: Invertigate how Mission Planner knows about each Mode name

Georacer, 2017-02-17:
Heartbeat message sends a numeric mode designator but Mission Planner knows the translation. Where does it pull the data from? What is the "M" variable type?

agentdoubleohpi, 2021-04-08:
Greetings,

The mapping for these modes are pulled from the MAVLINK [ardupilotmega.xml](https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/ardupilotmega.xml) definition. Not every mode is applicable to every platform, but it looks like the 'common' ones have identical numbers.

Georacer, 2021-04-08:
You have a point. By knowing which platform generated the log and with a bit of string manipulation, one could use [these enums](https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/ardupilotmega.xml#L950) to get the mode names.

Georacer, 2021-04-08:
So the follow-up question is how these mode names can be useful for this project. The lowest hanging fruit would be to make a list of the modes used in a log at the log information section.

AndersonRayner, 2021-04-08:
The modes are encoded from the 'mode.h' header in each of the vehicle types of APM. 
- (COPTER) https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/mode.h#L14
- (PLANE) https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/mode.h#L21
- (ROVER)  https://github.com/ArduPilot/ardupilot/blob/master/Rover/mode.h#L21

I made a couple of functions that convert the modes for me (hard-coded), but I don't think this information is stored in the .bin logs.  Below is the code I use that can return the mode based on `log.MODE.Mode` (as part of processing scripts I use one the data is imported via Ardupilog().  Unfortunately, as it is hard-coded, it would need to be updated in the future if the APM devs ever change the mode headers.

```
function modeName = modes_ArduCopter(modeNumber)
% From https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/mode.h#L14

switch (modeNumber)
    case  0; modeName = 'STABILIZE';
    case  1; modeName = 'ACRO';
    case  2; modeName = 'ALT_HOLD';
    case  3; modeName = 'AUTO';
    case  4; modeName = 'GUIDED';
    case  5; modeName = 'LOITER';
    case  6; modeName = 'RTL';
    case  7; modeName = 'CIRCLE';
    case  9; modeName = 'LAND';
    case 11; modeName = 'DRIFT';
    case 13; modeName = 'SPORT';
    case 14; modeName = 'FLIP';
    case 15; modeName = 'AUTOTUNE';
    case 16; modeName = 'POSHOLD';
    case 17; modeName = 'BRAKE';
    case 18; modeName = 'THROW';
    case 19; modeName = 'AVOID_ADSB';
    case 20; modeName = 'GUIDED_NOGPS';
    case 21; modeName = 'SMART_RTL';
    case 22; modeName = 'FLOWHOLD';
    case 23; modeName = 'FOLLOW';
    case 24; modeName = 'ZIGZAG';
    case 25; modeName = 'SYSTEMID';
    case 26; modeName = 'AUTOROTATE';
    otherwise; modeName = ['UNKNOWN_(',num2str(modeNumber),')'];
end
return
end

function modeName = modes_ArduPlane(modeNumber)
% from https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/mode.h#L21

switch (modeNumber)
    case  0; modeName = 'MANUAL';
    case  1; modeName = 'CIRCLE';
    case  2; modeName = 'STABILIZE';
    case  3; modeName = 'TRAINING';
    case  4; modeName = 'ACRO';
    case  5; modeName = 'FLY_BY_WIRE_A';
    case  6; modeName = 'FLY_BY_WIRE_B';
    case  7; modeName = 'CRUISE';
    case  8; modeName = 'AUTOTUNE';
    case 10; modeName = 'AUTO';
    case 11; modeName = 'RTL';
    case 12; modeName = 'LOITER';
    case 13; modeName = 'TAKEOFF';
    case 14; modeName = 'AVOID_ADSB';
    case 15; modeName = 'GUIDED';
    case 16; modeName = 'INITIALISING';
    case 17; modeName = 'QSTABILIZE';
    case 18; modeName = 'QHOVER';
    case 19; modeName = 'QLOITER';
    case 20; modeName = 'QLAND';
    case 21; modeName = 'QRTL';
    case 22; modeName = 'QAUTOTUNE';
    case 23; modeName = 'QACRO';
    case 24; modeName = 'THERMAL';
    otherwise; modeName = ['UNKNOWN_(',num2str(modeNumber),')'];
end
return
end

function modeName = modes_ArduRover(modeNumber)
% from https://github.com/ArduPilot/ardupilot/blob/master/Rover/mode.h#L21

switch (modeNumber)
    case  0; modeName = 'MANUAL';
    case  1; modeName = 'ACRO';
    case  3; modeName = 'STEERING';
    case  4; modeName = 'HOLD';
    case  5; modeName = 'LOITER';
    case  6; modeName = 'FOLLOW';
    case  7; modeName = 'SIMPLE';
    case 10; modeName = 'AUTO';
    case 11; modeName = 'RTL';
    case 12; modeName = 'SMART_RTL';
    case 15; modeName = 'GUIDED';
    case 16; modeName = 'INITIALISING';
    otherwise; modeName = ['UNKNOWN_(',num2str(modeNumber),')'];
end
return
end
```