Georacer/ardupilog #16: Apply message filters

Georacer, 2017-03-08:
Hunter,

Do you think it would be of any merit to filter the parsed messages before storing them, against a provided message list?

I think the complexity to do it during parsing would be substantial, whereas a simple array deletion post-parsing is trivial.
Plus, I don't expect any time savings if we choose to do it the 1st way.

hunt0r, 2017-03-08:
I'm not sure what you mean by "filtering" the parsed messages?

Georacer, 2017-03-08:
In case someone is interest only in e.g. RCOU, RPM, ATT messages, only these would be included in the output log.

Superficial, really, it's just that other similar analyzers provide that functionality.

hunt0r, 2017-03-08:
Oh yes, okay. I'll keep this in mind. It would make good sense to implement this with a move to a batch-style processing of LogMsgGroups, as mentioned in issue #9

Georacer, 2017-03-29:
Closed by #26

closed 2017-03-29 by Georacer