Georacer/ardupilog #17: Parsing large logs

Georacer, 2017-03-08:
I just tried to convert a large .BIN file (just under 4M messages). It was the log file from CanberraUAV's 2016 OBC run.

After an hour and a half, it had reached halfway, but it seemed like the progress rate would slow down progressively.

I fear that dynamic size allocation gets slower, as array size grows, which is understandable.

We should investigate the possibility to pre-allocate message arrays, to increase storage speed.

hunt0r, 2017-03-08:
Try the other branch in the repo, preallocate_msggroup, and let me know if that's better?

Note that at present, line 180 in Ardupilog.m sets the pre-allocation number for EVERY message type at 1/10th the number of TOTAL log headers. **This is a hack**, as I haven't pursued intelligent choices yet. Set it to something reasonable for your log and memory availability.

Depending on what you learn from this, we might merge this discussion with the broader topic of increasing total speed. (In that framework, I concluded preallocation seemed not to make a difference, but maybe you'll find that it does.)

Georacer, 2017-03-08:
Closing this and continuing in #9

closed 2017-03-08 by Georacer