Georacer/ardupilog #24: What's the difference between the "length" of a message as provided by FMT, vs the sum of the lengths of the field identifiers? (BQNnz, for example)

Georacer, 2017-03-26:
As extracted from code comments.

Georacer, 2017-03-26:
@hunt0r 
Do you think these may differ? Do you feel like we should check this for all messages?

hunt0r, 2017-03-27:
Well... I hope they don't.

Looking in LogStructure.h, it seems that the length provided by FMT is the returned value for sizeof() on the struct. Inside the struct definition, the field identifier string (e.g. "BQNnz") is hand-coded to correspond to the data in the struct.

So if they ever differ, I think that's a mistake. This seems like redundant information (which is to-be-avoided, in my opinion) but probably not worth our time.

I think we should assume they will always agree. Is that fine with you?

Georacer, 2017-03-27:
Let's leave this open and at some point one of us might code an assertion function for that.

Georacer, 2017-03-29:
This is tested by #31

closed 2017-03-29 by Georacer