Georacer/ardupilog #32: Remove (empty) LogMsgGroup if filtered out?

hunt0r, 2017-04-03:
With the filter capability (as added in #26) it makes sense to me to delete the (empty) LogMsgGroup if it was filtered out by the user. Do you agree?

I began working on it myself, but found it's a little more complicated than I thought. Just calling "delete(obj.(msgName))" deletes the object, but doesn't get rid of the property. I haven't yet figured out an easy way to clear it completely...

hunt0r, 2017-04-03:
When we filter, we might also want to delete the FMT data, I think it is saved before the filtering is applied.

Georacer, 2017-04-03:
Regarding deleting unwanted LogMsgGroups, including FMT, I'm for it.

Regarding the location of findInfo(), I placed it there to contrast again readLog. In my head, readLog is a dumb cataloguing and distribution of the content in the correct LogMsgGroups, whereas findInfo() is about meta-information. I don't have a strong opinion on the subject, I just don't want to overbloat readLog() without reason.

Georacer, 2017-04-27:
It doesn't look like property handles can be deleted (sigh).
How about we copy the structure and keep only the non-empy properties?