Georacer/ardupilog #35: Slicing functionality

Georacer, 2017-04-04:
It is desirable that functionality is implemented for:

- [ ] Cropping an Ardupilog object to a smaller time window
- [ ] Cropping one or more windows out of the log and keeping the rest

Arguments for cropping could be time or line indexed. For example:
getSlice([50 5000], ‘LineNo’)
getSlice([0 5.37],’TimeUS’)
getSlice([0 105.1],’TimeS’)
getSlice([t1 t2],’TimeUTC’)
double value = getSlice(5.4,’TimeS’) % For interpolation
getSlice(times,’type’,’endpoints’,’format’,’UTC’)
double interps = getSlice(times,’type’,’interp’,’format’,’TimeS’)

hunt0r, 2017-04-26:
Began addressing in #47

Georacer, 2018-10-06:
'TimeUS' was dropped, since not all messages have this field and also such fine-grained resolution is not needed.

hunt0r, 2019-02-07:
At present, I use slicing often, and it works pretty well. Should we close this Issue?

Georacer, 2019-02-08:
Yeah, I think it serves its purpose well. Go ahead!

closed 2019-02-08 by hunt0r