Georacer/ardupilog #38: Separate Units structure

Georacer, 2017-04-04:
Unit information is not embedded in the DF log currently.
Still, we might need those to effectively plot data ( #37 ).

A solution would be to upkeep a separate function or class which would hold the unit information for every field of every message possible.
This is a lot of manual work, but no other way exists currently. Even more work of one wants to make it work across versions.

Peter Barker mentioned here that he is has an addition of units in the DF in the works: https://github.com/ArduPilot/ardupilot/issues/5864

AndersonRayner, 2018-12-30:
Unit data (along with the multiplier) is now embedded in the DF log for many (if not all) of the log message types.  They're defined for each message type using the FMTU message which defines them based upon the message ID ( "FMTU", "QBNN", "TimeUS,FmtType,UnitIds,MultIds","s---", "F---" }.  

I think all the information you'll need to implement this one is sitting in https://github.com/ArduPilot/ardupilot/blob/master/libraries/DataFlash/LogStructure.h (in particular line 78-114 and line 1196).

Georacer, 2018-12-30:
Thanks for the heads up! I'll take a look at it.

Georacer, 2019-04-07:
#77 now uses this unit and multiplier information!

Georacer, 2019-04-16:
Closed by https://github.com/Georacer/ardupilog/pull/77

closed 2019-04-16 by Georacer