Georacer/ardupilog #39: Extract a bare-bones struct

Georacer, 2017-04-04:
Upon request we should be able to produce a bare bones MATLAB struct with all of the basic information.

Example:
myData:
 GPS:
  TimeUS: 1x44459 double
  Lat: 1x44459 double
  Lng:
 AHRS:
 BARO:

We don't like how Mission Planner does things currently, so we proposed our definition.

A good reason why this is needed is to help users share data with other people who don't have the Ardupilog source but want access to the log inside MATLAB.

hunt0r, 2017-04-26:
This was successfully addressed by #50.

On the same idea-thread: Do we want the capability to (re)create an Ardupilog object from one of these structs?

Georacer, 2017-04-26:
Nah, I don't think that's important. Just re-parse the original log (it fast ;) ).

Do you have any interesting scenaria?

hunt0r, 2017-04-26:
Haha, my head (heart?) got ahead of the game.

I'm dreaming that one day, people will post and share matlab-struct versions of dataflash and tlogs rather than the raw binary files... and everyone will want to use Ardupilog to read-and-write the structs, plot and visualize the data, and so forth. :)

closed 2017-04-26 by hunt0r