Georacer/ardupilog #45: Discussion: Code re-organization

hunt0r, 2017-04-04:
This idea got discussed alongside PR #41 which was discussed, and closed. I couldn't tell if you ever responded to this one directly? (My apologies if you already did) So I'm re-posting it as an issue.

[Code Design: or "What should we do with and without message filtering"]
I agree there's a design-choice to be made here, with pros-and-cons for any choice.

With the log parsing speed so fast, I propose the following general design:
1) Read the entire file into memory (a single call to fread).
2) Find all FMT messages.
3) Process them (at least) to extract the lengths and ID's of other message types which may exist.
3.5) At this point, there's probably no significant increase in time to also get also their names, fields, format strings, etc... but if we know the message will be filtered, we could theoretically skip this.
3.75) I think at this point, it also makes sense to search for the MSG type, and extract the firmware version, etc. (I think this is the findInfo() method, presently)
4) For FMT message defining an id/length, locate all valid messages of this type.

At this intermediate point, we have everything we need: the position of all valid messages in the log. We could also check for any regions of the log which are NOT part of a valid message, as well as any overlap. (if that's even possible) We also have a count of the total number of log lines, the number of lines of each type, etc.

5) For each message that was not filtered out, create a LogMsgGroup (if we didn't already do it in step3) and store the data. (Adding LineNo info wherever appropriate) Note that "storing the data" may get more complex in the near future, as we determine units, interpolability, etc. If we did create the LogMsgGroup back in line 3, I think at this point we agree that we should delete it for messages which are filtered out.

A con of this design is that we spend time finding (and counting) messages that we intend to filter, and could have skipped. Note that we still don't spend time processing-and-storing those messages, so the message filtering is still valuable for increasing speed, just not quite as fast as possible.

I hope the pros of this design outweigh the cons. Here are a few pros I see: having true LineNo info, having a count of the total log messages, having the ability to see if the log has lots of undefined/invalid messages (not presently implemented)

What do you think?

Georacer, 2017-04-04:
I had answered the following:

I run a profiler on the 150MB log on your latest version.
The delays are primarily split around
- line 168 (msg_LineNo = LineNo_vec(ismember(LineNo_ndx_vec, obj.valid_msgheader_cell{i,2}));)
- and 195-196, which is step 3.

In both cases we might be able to pull one or two tricks to make it faster. More specifically, I still don't understand what 168 does exactly.

But for the time being I agree in biting the bullet and going for it. It's orders of magnitude better than what's out there anyway.

Georacer, 2017-04-04:
I get it now. You compare the log indices of all msg headers with the known log indices of each message.
The result forms the vector of the log line number.

hunt0r, 2017-04-25:
I chatted with an experienced programmer, and it looks like I tend to subscribe to the philosophy from The Pragmatic Programmer (https://pragprog.com/book/tpp/the-pragmatic-programmer)

Aspects relevant to our discussion:
1) The key principle is "DRY: Don't Repeat Yourself"  So if something is "repeated" in the code, it exists twice and should be condensed into a single source. (e.g. made into a separate function)
2) Tracer bullet code: Often when designing/developing a program, it's best to code something, see how it works, then re-code (or trash it) as you discover better your use-cases and functionality needs.
3) Re-factoring code (what I've been calling re-organizing) is a common and appropriate part of the coding cycle/process. Do it as often as you think it needs to be done.

hunt0r, 2017-04-25:
From thinking about refactoring our code, I came to a discussion point: If we don't filter messages, the order that we do things doesn't matter too much, because we're doing them all anyways. If we DO filter messages, the order matters, because we need to balance tradeoffs between doing unnecessary processing and having nicer code.

So the design decisions can be re-stated as: How do we want message filtering to work?

Here's a 1st design decision: Do we want info filtered out of logs to be neglected everywhere, or only not recorded in a LogMsgGroup?  (Example: the MSG data carries the firmware version... if we filter out MSG, should the firmware version be blank?)

Here's a 2nd: If we want the log properties (e.g. firmware version) which come from filtered data, do we want to code "special-case" extraction of this data (e.g. firmware from MSG) or instead process it in the usual way (as if we were storing in a LogMsgGroup) but then NOT store the result?

Georacer, 2017-04-25:
I think you caught one of the few exceptions: message types which provide metadata.
I like the way we do things now: We parse enough to know what's there (the total amount of messages, firmware version and type), but don't store the result.

Storing is one of the most time-consuming processes anyway. The other is the first log scan for headers.

hunt0r, 2017-04-25:
Alright, I think we can proceed like that. I'll wait till we resolve the outstanding pull requests before implementing this (re)design.

Next design idea: Since we're already finding all valid message data, even if we don't store it, do we want to keep any record of when data is used as part of a valid message?

This would enable us to find any unused portions of the log (incomplete messages? undefined messages?) as well as if we ever doubly-use any bytes in two distinct messages. (Very unlikely to happen, I think?)

If we're keeping record of unused log data during processing, this could also speed up searching for next data: Just search the portion of the log that hasn't already been used, not re-searching the whole log.

Georacer, 2017-04-25:
Checking for unused parts of the log may happen, but I think it's not a priority. We have other more important plans already.
Checking for doubly-used parts of the log might be an overkill methinks.
Both may require significant size overhead.

Recording the unused parts of the log is not required for further searches: The whole log is scanned only **once** for the header bytes (discoverHeaders). Afterwards we lookup only at the known header locations for valid messages (which are significantly fewer).

If you want to speed up last part, you could do that, but I don't think there's much merit in doing it.

closed 2018-10-10 by hunt0r