Georacer/ardupilog #58: Bug: which() doesn't work as desired

hunt0r, 2017-09-01:
At the beginning of ArduPilog, if a filename is specified as the .BIN file, the InputParser should detect that as p.Results.path and call fileparts(which(p.Results.path)). (Line 47 of ArduPilog.m, I believe)

Unfortunately, which() doesn't seem to work as desired with files specified as 'path/to/file.ext'

I can demonstrate this at the command line. Having a valid file and typing:
which('valid/path/to/valid_file.ext')
returns:
'valid/path/to/valid_file.ext' not found.  (<- this is incorrect)

While exist('valid/path/to/valid_file.ext') returns 2, indicating that a file is found and exists.

If I addpath('valid/path/to/') and then which('valid_file.ext') the function works as desired, returning:
valid/path/to/valid_file.ext   (<- as a MATLAB string)

I'm going to hack a quick workaround for now, but we should decide in the future some better fix.

Georacer, 2018-10-06:
Probably matlab only acknowledges files that are inside its current path (i.e. scope). Not sure why `exists` works, but this isn't the only time Matlab is inconsisten.t

hunt0r, 2018-10-08:
Commit #62 commented out the call to MATLAB's offending which(), so I think we can declare this issue closed.

closed 2018-10-08 by hunt0r