Georacer/ardupilog #59: SBRH and SBRM have number-only fields, causes MATLAB warning

hunt0r, 2017-10-23:
Looks like the commit referenced below for LogStructures.h set the SBRH and SBRM fields to integers 1,2,3,4,... and MATLAB  complains with the warning linked below.

For a first work-around, we could change SBRM 1 to "m1", 2 to "m2", and so on... probably SBRH similarly with 1 = "h1", 2 = "h2", 3 = "h3", and so on.

[The MATLAB warning]

Warning: '1' is not a valid dynamic property name.  Property names must begin
with a letter and be composed of letters, numbers, and underscores.  This will
become an error in a future release. 
> In LogMsgGroup/storeFormat (line 35)
  In LogMsgGroup (line 24)
  In Ardupilog/createLogMsgGroups (line 304)
  In Ardupilog/readLog (line 128)
  In Ardupilog (line 62)

[The git commit where this probably changed]
commit 0b5dad33f26b96ffaed212d75674c30dfa07f422
Author: ebethon <etienne@swiftnav.com>
Date:   Mon Jun 26 16:41:24 2017 +0100

    DataFlash: Rename SBR1/SBR2 to SBRH/SBRM and change format
    
    Add (sender_id, msg_len) in SBRM message
    Add index/pages in SBRH/SBRM messages
    Change format to integers to avoid GCS string parsing
    Decrease data size in SBRH

Georacer, 2017-10-23:
Actually, I think I already have a fix for that. Hang on a sec.

Georacer, 2018-10-06:
Closed with #60

closed 2018-10-06 by Georacer