Georacer/ardupilog #68: Release blog on Ardupilot blog

Georacer, 2018-10-06:
@hunt0r Here's a proposal for the release blog. All suggestions welcome.

> Hello everyone!
> 1.5 year ago, me and @hunt0r needed to parse some ArduPlane logs. We are both on our track to get a PhD in UAV control systems and Matlab is a very common tool in our trade.
>
> Sadly, at the time the only tool able to convert from DataFlash logs (.bin) to a format readable by Matlab was Mission Planner and it didn't do a good job at that. It would work only on Windows, take a very long time to convert and would ignore all text fields.
>
> So, we decided to create our own converter, `ardupilog`. It is Matlab-based, meaning that you need Matlab to run it.
> It can read all DataFlash logs, since they are self-defining, and create a versatile log object.
> You can even "flatten" the log into a classic struct, if you want the information to be independent of our library.
>
> # Example Usage
>
> ## Reading a log
> Add the `ardupilog` source code to your path.
> Then,
> ```matlab
> log = Ardupilog()
> ```
> will open a file browser, where you can select the log file you want to decode.
> The variable struct `log` will be generated with the included message types as fields.
> Its contents can be normally processed with Matlab commands:
> << Placeholder for images of the resulting struct and a plot >>
>
> ## Filtering messages
> You can optionally filter the log file for specific message types:
> ```matlab
> log = Ardupilog('<path-to-log', <msgFilter>)
> ```
> `msgFilter` can be:
> * Either a vector of integers, representing the message IDs you want to convert.
> * Or a cell array of strings. Each string is the literal name of the message type.
> << Placeholder image with filtered log >>
> 
> ## Slicing
> Typically, only a small portion of the flight log is of interest. Ardupilog supports *slicing* logs to a specific start-end interval. For example, to keep log messages from boot time 1000sec to 2500sec:
> ```matlab
> sliced_log = log.getSlice([1000, 2500], 'TimeS')
> ```
> << Placeholder image with sliced log >>
>
> ## Exporting to plain struct
> Creating a more basic struct file, free of the `ardupilog` dependency, is possible with:
> ```matlab
> log_struct = log.getStruct();
> ```
> `log_struct` does not need the `ardupilog` source code accompanying it to be shared.
> << Placeholder image with plain struct >>
>
> More information can be found on the repo's README.

Georacer, 2018-10-06:
Do we want to have a uniform format for the Ardupilog name?
Possible options:
* Ardupilog
* ardupilog
* _ardupilog_ (in italics)
* `ardupilog` (in true-type)
Is anyone of your fancy?

hunt0r, 2018-10-08:
For the name: I slightly prefer it capitalized (coming from MATLAB's convention to capitalize class names, I think) and I think the true-type looks nice too, if that's easy. But I'm okay with something else if you like, it doesn't matter much to me.

For the blog-post announcement, I noticed a few tiny grammar things that I would change:
1) I prefer not starting sentences with numbers, so I'd replace ("1.5 year ago, ...") with either "About a year ago ..." or maybe "About two years ago..." If you like to be more specific on the time, "A year-and-a-half ago, ..."
2) Two minor grammar formalities: When listing other people and yourself, it's considered proper to list yourself last. (So change "me and @hunt0r" to "@hunt0r and me")  But also, since we are the subjects of the sentence (not the objects) you should use "I" instead of "me".  **So I propose changing to "A year-and-a-half ago, @hunt0r and I needed to parse some ArduPlane logs..."**
3) When talking about Mission Planner, can we avoid the possibility of insult by changing "it [Mission Planner] didn't do a good job" to something like "it didn't serve our needs well enough."
4) Is this a good place to mention a "wishlist" item that someone else might code for us? The major one that comes to mind: I wish Ardupilog could also encode .tlog files to a similar MATLAB format. But I'm not willing to implement it myself.

Overall, it looks great! What else might I do to help?

Georacer, 2018-10-08:
Posted here: https://discuss.ardupilot.org/t/ardupilog-a-dataflash-log-to-matlab-converter/33812
Close this if no more edits are needed.

hunt0r, 2018-10-08:
It looks outstanding. Thanks so much for that!

closed 2018-10-08 by hunt0r