Georacer/ardupilog #69: Logs in root file not visible

Georacer, 2018-10-08:
Entering
```matlab
log = Ardupilog('mylog.bin')
```
will error out.
The problem is probably due to
https://github.com/Georacer/ardupilog/blob/7fda3fa215da2b29c207bf1ba1eeb352ce77bf3e/Ardupilog.m#L106
because `filePathName` is empty.

```matlab
log = Ardupilog('./mylog.bin')
```
works.

hunt0r, 2018-10-08:
The syntax you linked:
`log = Ardupilog('mylog.bin')`
runs WITHOUT error for me. I'm on version R2015a, and I have access to several other versions (2016a, 2017a, 2018a, I think). Should I try it on those?
my obj.filePathName == '' which is the empty string.

Georacer, 2018-10-08:
Running R2017a here. Can you give it a shot?

hunt0r, 2018-10-08:
In 2016a, it runs successfully.
In 2017a and 2018a I get the same error as you (fopen must not like the empty-string path).

So I agree we should proceed to fix. (Maybe just if isempty(path); path = '.'; end; or something quick like that?)

hunt0r, 2018-10-08:
BTW, I hope it doesn't matter, but
R2015a on linux, ran successfully.
R2016a on Windows, ran successfully.
R2017a on Windows, error occurred. (using './mylog.bin' works)
R2018a on Windows, error occurred. (using './mylog.bin' works)