Georacer/ardupilog #78: Issues with parsing ChibiOS based ardupilot logs

vbnhu, 2019-09-16:
The Matlab script worked flawlessly for a while, but now they switched to Chibios based operating system and possibly changed the log description, so ardupilog gives an error.
A file to test is attached here:
https://wetransfer.com/downloads/72b68488cf91076b94ba4750dc2f48be20190916175750/52c55b5efd8eb26dbc0d4b010883e7c220190916175750/fe0f05

hunt0r, 2019-09-17:
I regret that I don't have the ability to continue supporting Ardupilog. I hope you (or someone) are willing to take on the task!

Georacer, 2019-09-17:
I'll take a look. Thanks for the log!

Georacer, 2019-09-21:
So... With the current master and no code changes I can parse your log fine.
Here's a Pitch plot from the AHR2 message.
[image]

Are you sure you are not missing a step somewhere?

The way I'm loading the log is placing it in a `logs` subfolder and enter
`log = Arupilog('logs/logName.BIN')`
Keep in mind that the file names are case sensitive!

Georacer, 2019-09-24:
@vbnhu Did you manage to parse your logs in the end?

vbnhu, 2019-09-25:
> @vbnhu Did you manage to parse your logs in the end?

Thanks, indeed it worked with this file. Possibly things got fixed in 3.9.9 and I used a development version of the Ardupilog.m from my own computer which still errored out, but when I downloaded the version from Github it looked OK. I think there is no need to look at previous versions where I started to investigate the error, since the Ardupilot team fixed the issue. If I run into it again I will post a proper log. Thanks again!

closed 2019-10-04 by Georacer