Georacer/ardupilog #79: Log not parsing - throws error

Georacer, 2020-08-12:
According to this report: https://discuss.ardupilot.org/t/ardupilog-a-dataflash-log-to-matlab-converter/33812/11?u=georacer

Reportedly it's from (Arduplane?) 3.9.7.
[testlog.zip](https://github.com/Georacer/ardupilog/files/5065738/testlog.zip)

Log file attached.

Georacer, 2020-08-12:
This is the list of the formats found in the log:

```
    [128]    'FMT'     [ 89]
    [202]    'UNIT'    [ 76]
    [201]    'FMTU'    [ 44]
    [203]    'MULT'    [ 20]
    [129]    'PARM'    [ 31]
    [130]    'GPS'     [ 46]
    [131]    'GPS2'    [ 46]
    [132]    'GPSB'    [ 46]
    [195]    'GPA'     [ 26]
    [196]    'GPA2'    [ 26]
    [197]    'GPAB'    [ 26]
    [133]    'IMU'     [ 53]
    [134]    'MSG'     [ 75]
    [135]    'RCIN'    [ 39]
    [136]    'RCOU'    [ 39]
    [137]    'RSSI'    [ 15]
    [139]    'BARO'    [ 37]
    [140]    'POWR'    [ 22]
    [143]    'CMD'     [ 46]
    [144]    'RAD'     [ 20]
    [146]    'CAM'     [ 43]
    [212]    'TRIG'    [ 43]
    [162]    'ARSP'    [ 32]
    [226]    'ASP2'    [ 32]
    [164]    'BAT'     [ 37]
    [165]    'BAT2'    [ 37]
    [166]    'BCL'     [ 35]
    [167]    'BCL2'    [ 35]
    [163]    'ATT'     [ 27]
    [168]    'MAG'     [ 34]
    [171]    'MODE'    [ 14]
    [198]    'RFND'    [ 19]
    [200]    'DMS'     [ 33]
    [220]    'BCN'     [ 41]
    [221]    'PRX'     [ 56]
    [227]    'PM'      [ 25]
    [223]    'SRTL'    [ 29]
    [138]    'IMU2'    [ 53]
    [147]    'IMU3'    [ 53]
    [141]    'AHR2'    [ 45]
    [182]    'POS'     [ 31]
    [142]    'SIM'     [ 45]
    [ 64]    'NKF1'    [ 55]
    [ 65]    'NKF2'    [ 35]
    [ 66]    'NKF3'    [ 33]
    [ 67]    'NKF4'    [ 35]
    [ 68]    'NKF5'    [ 40]
    [ 69]    'NKF6'    [ 55]
    [ 70]    'NKF7'    [ 35]
    [ 71]    'NKF8'    [ 33]
    [ 72]    'NKF9'    [ 35]
    [ 73]    'NKF0'    [ 36]
    [ 74]    'NKQ1'    [ 27]
    [ 75]    'NKQ2'    [ 27]
    [ 76]    'XKF1'    [ 55]
    [ 77]    'XKF2'    [ 34]
    [ 78]    'XKF3'    [ 33]
    [ 79]    'XKF4'    [ 35]
    [ 80]    'XKF5'    [ 40]
    [ 81]    'XKF6'    [ 55]
    [ 82]    'XKF7'    [ 34]
    [ 83]    'XKF8'    [ 33]
    [ 84]    'XKF9'    [ 35]
    [ 85]    'XKF0'    [ 36]
    [ 86]    'XKQ1'    [ 27]
    [ 87]    'XKQ2'    [ 27]
    [ 88]    'XKFD'    [ 35]
    [ 89]    'XKV1'    [ 59]
    [ 90]    'XKV2'    [ 59]
    [148]    'TERR'    [ 34]
    [149]    'UBX1'    [ 22]
    [150]    'UBX2'    [ 16]
    [151]    'UBY1'    [ 22]
    [152]    'UBY2'    [ 16]
    [172]    'GRAW'    [ 42]
    [173]    'GRXH'    [ 24]
    [174]    'GRXS'    [ 41]
    [175]    'SBFE'    [ 63]
    [153]    'ESC1'    [ 23]
    [154]    'ESC2'    [ 23]
    [155]    'ESC3'    [ 23]
    [156]    'ESC4'    [ 23]
    [157]    'ESC5'    [ 23]
    [158]    'ESC6'    [ 23]
    [159]    'ESC7'    [ 23]
    [160]    'ESC8'    [ 23]
    [169]    'MAG2'    [ 34]
    [170]    'MAG3'    [ 34]
    [176]    'ACC1'    [ 31]
    [177]    'ACC2'    [ 31]
    [178]    'ACC3'    [ 31]
    [179]    'GYR1'    [ 31]
    [180]    'GYR2'    [ 31]
    [181]    'GYR3'    [ 31]
    [183]    'PIDR'    [ 35]
    [184]    'PIDP'    [ 35]
    [185]    'PIDY'    [ 35]
    [186]    'PIDA'    [ 35]
    [187]    'PIDS'    [ 35]
    [188]    'DSTL'    [ 56]
    [161]    'BAR2'    [ 37]
    [199]    'BAR3'    [ 37]
    [189]    'VIBE'    [ 35]
    [190]    'IMT'     [ 47]
    [191]    'IMT2'    [ 47]
    [192]    'IMT3'    [ 47]
    [224]    'ISBH'    [ 31]
    [225]    'ISBD'    [207]
    [193]    'ORGN'    [ 24]
    [222]    'DSF'     [ 34]
    [194]    'RPM'     [ 19]
    [213]    'GMB1'    [ 47]
    [214]    'GMB2'    [ 44]
    [215]    'GMB3'    [ 13]
    [216]    'RATE'    [ 59]
    [217]    'RALY'    [ 23]
    [218]    'VISO'    [ 43]
    [204]    'SBPH'    [ 23]
    [209]    'SBRH'    [ 67]
    [210]    'SBRM'    [123]
    [211]    'SBRE'    [ 23]
    [ 11]    'ATUN'    [ 41]
    [ 12]    'ATDE'    [ 19]
    [ 14]    'PTUN'    [ 22]
    [  3]    'OF'      [ 28]
    [  2]    'CTUN'    [ 53]
    [ 13]    'MOTB'    [ 27]
    [  4]    'EV'      [ 12]
    [  6]    'D16'     [ 14]
    [  7]    'DU16'    [ 14]
    [  8]    'D32'     [ 16]
    [  9]    'DU32'    [ 16]
    [ 10]    'DFLT'    [ 16]
    [  5]    'ERR'     [ 13]
    [ 16]    'PL'      [ 41]
    [ 17]    'GUID'    [ 36]
    [253]    'FTSS'    [ 24]
    [252]    'FTS2'    [ 14]
    [248]    'NKT1'    [ 47]
    [247]    'NKT2'    [ 47]
    [246]    'PSC'     [ 59]
    [245]    'CTRL'    [ 31]
    [254]    'CSTA'    [ 21]
    [251]    'FTSV'    [ 27]
    [250]    'GPST'    [ 28]
    [249]    'LOSM'    [ 23]
```

Weirdly enough, LogStructure.h from master (dates 20d ago) doesn't list many of the above fields (e.g. LOSM).

On the other hand, the commit tagged Arduplane-3.9.7 (dates mid 2019) has its dataflash library in a different place, not in AP_Logger. Nonetheless, it doesn't contain mentions to these messages either.

Georacer, 2020-08-12:
I think I'm getting closer. The log doesn't have any UNIT and MULT messages, but it does have ONE(?) FMTU message. This makes it drop into the piece of code which tires to decode units information and fails.
Will patch with code checking the number of unit-related messages.

sharmasnjv, 2020-08-12:
Thanks! I do get the same error on this file. I got the uploaded error On a different file.

Georacer, 2020-08-12:
Should work now, please pull the new master.

closed 2020-08-12 by Georacer

sharmasnjv, 2020-08-13:
Thanks. Confirmed that the fix worked. This code is a modified branch.

Georacer, 2020-08-13:
> Thanks. Confirmed that the fix worked. This code is a modified branch.

Glad that it worked. Cheers!

commit 3b64a165 (2020-08-12): Handled malformed unit-related messages

Will now issue warning if there is inadequate information to decode
message units.
Fixes #79.