Georacer/ardupilog #89: Error running the code

isso11, 2023-01-04:
Hi,

When I run the code and select a .BIN log  file, I get this error:

```
'RSSI' is already defined as a property.

Error in Ardupilog/createLogMsgGroups (line 322)
                    addprop(obj, newName);
```

I tried to upload the .BIN file on Flight Log Review website and it works fine. So I assume the file itself is okay.

Any help would be appreciated!

Georacer, 2023-01-04:
Hello! There is a very recent PR that might fix this new issue: https://github.com/Georacer/ardupilog/pull/88

I haven't merged it yet, I will do it within the next week. In the meantime, you might be able to use that branch to open your log, if you know how to do it.

isso11, 2023-01-04:
Thank you. It worked. @Georacer

closed 2023-01-04 by isso11