Georacer/ardupilog #91: ardupilog halts on finding (assumed) .BIN glitch

philatdunetive, 2023-08-30:
Hello 
I've been using ardupilog for a few years and very impressed. BUT Just tried to convert a 2 hour .BIN file (from ArduRover in boat mode) recorded yesterday and the script fails with these warning and errors: 
----------
Warning: incompatible data on msg type=0/  ð (>>>00<<< >>>00<<<)incompatible data on msg type=0/ð (>>> 
> In LogMsgGroup/storeFormat (line 44)
In LogMsgGroup (line 32)
In Ardupilog/createLogMsgGroups (line 323)
In Ardupilog/readLog (line 135)
In Ardupilog (line 68)
In pixhawkBinToLogViaCLASS_corruptFileAnalysis (line 44) 
Error using Ardupilog/addprop
'  ð' is not a valid dynamic property name. Property names must begin with a letter and be composed of letters, numbers, and underscores.

Error in Ardupilog/createLogMsgGroups (line 329)
                    addprop(obj, newName);

Error in Ardupilog/readLog (line 135)
            obj.createLogMsgGroups(data');

Error in Ardupilog (line 68)
            readLog(obj);

Error in pixhawkBinToLogViaCLASS_corruptFileAnalysis (line 44)
        log = Ardupilog(fNameOP); 

---
[I've added the hex output in the 1st line to check that the type_num and type_name are both zero (null, 0x00 etc)  ];

this is about 9,000,000 doubles along the file (72,000,000 bytes along): total file length 407,115,390 Bytes

My Matlab coding is not expert, so I am trying to get the code to just ignore the 'glitch' and catch up with the next valid record, but the structure of the script defeats my sledgehammer attempts. Frustrating is that the code stops at the error , and fails to write what it has thus far. 

Thoughts and debugs most welcome

Phil

Georacer, 2023-08-30:
Hi there!

I'm a bit perplexed by the line `In pixhawkBinToLogViaCLASS_corruptFileAnalysis (line 44) `.
Is this your own code?

It seems that there is a message declared in the log whose name (at the least) is non-ASCII (or basically mangled hex values): 0/�ð
You get the warning that Ardupilog can't store data for it, as it can't understand its format. This will fail gracefully.

But when Ardupilog tries to create a property in the log with that funky name, it can't: Matlab refuses to assign a non-ASCII name.
You could start by trying to have Ardupilot catch the error in `createLogMsgGroups`, around linke 323 (in my file, 329 in yours)
```matlab
                new_msg_group = LogMsgGroup(newType, newName, newLen, newFmt, newLabels);
                if isempty(new_msg_group)
                    warning('Msg group %d/%s could not be created', newType, newName);
                else
                    obj.msgsContained{end+1} = newName;
					try
						addprop(obj, newName);
                    catch ME
                        if strcmp(ME.identifier,'MATLAB:class:PropertyInUse')
                            warning('Duplicate message %d/%s definition', newType, newName);
                        else
                            rethrow(ME);
                        end
					end
                    obj.(newName) = new_msg_group;
                end
```
and then clean up whatever remains.

But perhaps the easiest for you would be for you to send me the log (e.g. via WeTransfer) so I can make a clean fix.

philatdunetive, 2023-08-30:
Hello,

Thanks for replying so quickly to the post: yep, the >>>pixhawkBinToLogViaCLASS_corruptFileAnalysis<<< is my simple script: it just reads in the .BIN,  does some crude slicing, and then calls ardupilog with a subset of the .BIN as a tool to see where error is: I'd assumed it was a frayed endOfFile... but no, it's buried in the middle.

Of interest: the Mission Planner 'review flight' reads the file OK. Not meant as a critique, but to confirm the BIN is mainly OK...

Back to getting an actual solution: I'll send the 0.3GB file if I may: could you provide suitable receive email for WeTransfer to use? Big thanks in advance

Georacer, 2023-08-30:
Is the e-mail in my profile visible? If yes, you can send it there. Otherwise, I think WeTransfer allows you to paste a link here. You can disable that link later.

philatdunetive, 2023-08-30:
Try 
https://we.tl/t-AZtQ59KrFg

Georacer, 2023-08-30:
I managed to parse your log. There seemed to be a FMT message that contained invalid (or corrupted) information.

I've set Ardupilog to ignore it and all other data that correspond to that message (I hope it not important ^^).

I'll push an update soon.

Georacer, 2023-08-30:
Check again the tag `v1.1.0` and see if completes the parsing. Let me know if it looks as if important messages are missing from the log.

philatdunetive, 2023-08-30:
Looks good: throws up Warnings but carries on and appears to complete full conversion. Will check for significant time-step jumps tomorrow. Good work!

Georacer, 2023-08-30:
Cheers!