Georacer/ardupilog #95: Data Resampling

Husseinhajjsleiman, 2025-02-05:
Greetings, 

Thank you for your work! I have been using ArduPilog for a while, and it has been really helpful.

I have one concern and would like to confirm it. I increased the sampling rate of IMU and GYR data through ArduCopter, and when I plot it in Mission Planner, I can see that I achieved the desired sampling rate. However, when I convert the BIN file to MAT using ArduPilog, the sampling rate becomes lower.
Is any resampling rate for the data being done during the conversion process?

Georacer, 2025-02-05:
Hello! I'm glad to hear you like the program.

Can you perhaps share the log that's giving you trouble?

Husseinhajjsleiman, 2025-02-05:
Thank you for your immediate response. Here is a [link ](https://drive.google.com/file/d/1s-ZWN5EXWWln5vkS65siEaOGbKVqqXXz/view?usp=sharing)for the bin file, which is uploaded on my google drive. 

Through mission planner i can see that the sampling rate of the IMU is much higher. I just want to understand if there's any resampling is happening when we convert the data? Is their a way to prevent that and get raw data?

Below is a picture shown how did i calculate the sampling rate of IMU and GYR through matlab.

[image]

Georacer, 2025-02-07:
I don't see the issue.

`IMU` has the low rate logging. `ACC` and `GYR` have the high rate logging.

I see the same result in Mission Planner and Ardupilog:

[image]

```matlab
>> 1/mean(diff(log.IMU.TimeS))

ans =

    7.5447

>> 1/mean(diff(log.ACC.TimeS))

ans =

   1.5892e+03

>> 1/mean(diff(log.GYR.TimeS))

ans =

   1.5742e+03
```

Husseinhajjsleiman, 2025-03-06:
I am sorry for my late reply. This was a mistake from my side looking at the wrong data set.
May I know why 'IMU' data have low sampling rate and the ACC have much higher? although I am assuming that the source of acceleration is from the IMU.. 
I know it might not be the right place to ask this but Im thankful for your time if you have any thoughts about it

Georacer, 2025-03-06:
`ACC` is specially logged raw accelerometer when you enable it in via `LOG_BITMASK` (IIRC). It's very high rate and used to tackle vibration issues.
`IMU` is your standard, run of the mill IMU logging.

closed 2025-03-06 by Georacer