Georacer/ardupilog #96: ESC data is not logged despite having it

Husseinhajjsleiman, 2025-03-07:
Hi, 

I have flight test data where I used sysID flight mode from arducopter. ESC are logged via telemetry. If i use the bin file in uav log viewer or mission planner i can see the data from ESC however, when i use Ardupilog only one ESC data is not shown. I tested the conversion with another flight data that have sysID flight mode initiated but the issue did not occur. Do you have an idea why only with this flight log the issue is occurring?

[image]

[image]

Georacer, 2025-03-08:
Hello @Husseinhajjsleiman ,

Would it be possible to send over the .bin log? Then I can debug the problem.

Georacer, 2025-03-15:
For me, your log shows a bunch more ESC instances!

[image]

Very weird that we're getting different results. Are you using the latest `master` of Ardupilog?

Georacer, 2025-03-15:
So, in my case, here are the instances found for this log:

[image]

Instances 10, 11, 12, 13 are filled with ~4300 datapoints each, which seems correct.

That means that I can't reproduce your error, but I have another error on my hands, with the extra instances and whatnot.

Husseinhajjsleiman, 2025-03-23:
It could be the reason because I have activated high sampling rate for IMU and at the same time i was logging for notch filtering. I have recently changed the sd card to a bigger one and that have higher write speed. Unfortunately, I was trying to replicate the issue with the older sd card but i could not. So im really not sure what exactly happened. Please suggest, If you have any idea for a test i can do to replicate it. Thank you again for the update!

Georacer, 2025-03-23:
I'm not worried about replicating this, to be honest.
We have this one example that says basically "If you get corrupt data with correct header, msgID and length", then it will be accepted.
There is little that one can do in case of data corruption, I don't think it's Ardupilog's job to clean it.

Although I'm thinking of imposing monotonic constraints on the timestamp. This would have helped in this case.